rust-iot/rust-radio-sx127x #7: Question: SPI settings Raspberry Pi 3 B+ Lora HAT

ganwell, 2020-05-12:
I am new to embedded, I hope it is ok to ask here, I have so much information and I can't figure out how it all relates. I am hoping to get some pointer that helps me to connect the information. I am trying to find out how the settings:

* cs pin
* rst pin
* busy pin

translate to the information from the Lora HAT [1] From their sample code that works and runs with WiringPI [2] I get these pin numbers:

```c
int ssPin = 6;
int dio0  = 7;
int RST   = 0;
```

* I learned that nss, ss, ssPin, ce (chip enable) and cs (chip select) is the same thing. So it got one pin number, but I don·'t know if the numbering is the same.

* I assume that RST is reset

* This leaves dio0 to be busy??

The names the lora HAT uses in documentation is

* Lora DIO0
* Lora RESET
* Lora NSS
* Lora SCK

The biggest problem is, every subsystem numbers the pins different. See image:

[image]

Does the kernel translate the pin numbers too? I mean we access SPI via kernel device.

I converted the numbers in many way from most reasonable to some strange ways. What ever I do there are two outcomes

* InvalidDevice (After that, the sx127x still works with the C program)
* ResourceBusy (The sx127x is bricked and I have to reboot)

[image]

I also read about SPI and naming conventions didn't completely match neither rust-radio-sx127x nor one used in the Lora HAT.

[1] http://wiki.dragino.com/index.php?title=Lora/GPS_HAT
[2] http://wiringpi.com/

ryankurte, 2020-05-13:
ahh yes this is a classic embedded adventure, sorry that i haven't documented the generics better (if you fancy a PR later we could add some better docs and instructions for this shield).

> I learned that nss, ss, ssPin, ce (chip enable) and cs (chip select) is the same thing.

there isn't super common nomenclature but yep, all talking about the SPI chip select pin

> I assume that RST is reset

yep

> This leaves dio0 to be busy??

yeah, the radio has a set of configurable `DIO` pins, at boot `DIO0` is a `BUSY` signal that lets you know whether the radio is, busy. Technically it's possible to use other pins for this, but practically this could probably be renamed to `DIO0` to be less ambiguous as there isn't a lot of reason to remap this.

> I converted the numbers in many way from most reasonable to some strange ways. What ever I do there are two outcomes

> InvalidDevice (After that, the sx127x still works with the C program) 

As in you can interact with the device and send radio messages and things with `sx127x-util`? Usually this means the configuration is wrong and you can't communicate with the device in which case nothing should work.

It's is however possible you have a different version of the IC and that [this](https://github.com/ryankurte/rust-radio-sx127x/blob/4ffe64e03aad9819d7ae76b69565250837e990ed/src/lib.rs#L160) needs updating to add the different version number.

It's also possible you haven't got the right reset pin mapped, because the driver will attempt to reboot the radio on connect / you shouldn't need to restart the pi.

> I also read about SPI and naming conventions didn't completely match neither rust-radio-sx127x nor one used in the Lora HAT.

So long as you've enabled the SPI / are using the right spidev you shouldn't need to worry about MOSI/MISO (this is defined by the SPI device, and works on my rpi), should be just `DIO=7`, `RST=0`, and `CS=6` from the table you linked. The pin numbers you're looking for (that the kernel understands) are the ones on the rpi gpio header, as in the image [here](https://www.raspberrypi-spy.co.uk/2012/06/simple-guide-to-the-rpi-gpio-header-and-pins/).

ganwell, 2020-05-17:
Thanks a lot for the help! Now I am quite sure the pins are correct. I attached a gdb to the program. It panics [here](https://github.com/ryankurte/rust-radio-sx127x/blob/4ffe64e03aad9819d7ae76b69565250837e990ed/src/util/main.rs#L66)

```
thread 'main' panicked at 'error exporting busy pin: Io(Os { code: 16, kind: Other, message: "Resource busy" })', src/libcore/result.rs:1165:5
```

When I have some spare time, I'll clone the dependencies so I can step into `export`.

I also made sure that I have IC version 0x12 using the C program.