rust-iot/rust-radio-sx127x #13: lora_check_receive  restart argument

pdgilbert, 2020-11-12:
Following code in the _tests/integration.rs_ file I have my `lora_sp_receive` example working on an stm32f411 but I am confused by the restart option, especial a `true` setting,  and not sure if I correctly understand the comment (src/lora.rs line 336)

    /// This returns true if a boolean indicating whether a packet has been received.
    /// The restart option specifies whether transient timeout or CRC errors should be
    /// internally handled (returning Ok(false)) or passed back to the caller as errors.
    
Does that mean that 
- `lora_check_receive(false)`  passes the check result back to user code and returns `Ok(true)` if a packet has been received and `Ok(false)` if nothing has been received or if there has been a  timeout or CRC error.
  
- `lora_check_receive(true)` always returns `Ok(false)`.  It handles timeouts and CRC errors without passing back to user code. It waits until something  is received?  So blocks?

Why is the argument called "restart option"? 

Which restart option do you suggest in what situations?

For reference, the example code is at https://github.com/pdgilbert/eg_stm_hal/blob/master/examples/lora_spi_receive.rs and currently the main part is

```
#[entry]
fn main() -> !{

    let mut lora =  setup();         //delay is available in lora
    
    lora.start_receive().unwrap();   // should handle error

    let mut buff = [0u8; 1024];
    let mut n: usize ;
    let mut info = PacketInfo::default();

    loop {

       let poll = lora.check_receive(false);    

       match poll {
            Ok(v)  if v  =>  {n = lora.get_received(&mut info, &mut buff).unwrap();
                              hprintln!("RX complete ({:?}, length: {})", info, n).unwrap();
                              //hprintln!("{:?}", &buff[..n]).unwrap();
                              hprintln!("{}", to_str(&buff[..n])).unwrap();
                              },

            Ok(_v)       =>  hprint!(".").unwrap(),           // print . if nothing received
            
	    Err(err)     =>  hprintln!("poll error {:?} ", err).unwrap(),
            };

       lora.delay_ms(100u32);
       };

}
```

ryankurte, 2020-11-12:
as you've determined you need to poll on `check_receive` to see whether something is received, all of this is just interacting with  the internal radio state machine. 

setting `restart` lets the driver internally deal with any CRC or recoverable errors and re-enter receive mode (if it can, otherwise you'll still get an error), not setting this returns errors to the higher level for handling but requires another `start_receive` call to return the radio to receive mode when an error occurs. you can see the code for this [here](https://github.com/rust-iot/rust-radio-sx127x/blob/2285c1ca9f527ab464a27f6a43af3b0a93130d2f/src/lora.rs#L371)

pdgilbert, 2020-11-13:
Ok, I think I understand. I guess there are not concepts of optional or default  arguments in Rust, and  `restart `is `bool` rather than `Option(bool)`,  so I suppose by "setting it" you mean setting in` true` and by "not setting it" you mean setting it `false`. (I realize this is a bit pedantic but I have a differ background.)

ryankurte, 2020-11-14:
yeah pretty much, it _could_ be an object with more stuff but, that's also adding another associated type which as you've discovered proportionately increases the complexity of integrating traits... a documentation improvement would be welcome if you can think of a better way of presenting this

pdgilbert, 2024-03-21:
This issue is out-of-date and effectively resolved.

closed 2024-03-21 by pdgilbert