rust-iot/rust-radio-sx127x #32: Type issue with stm32l1xx_hal

Niondir, 2021-02-14:
I try to use the driver with the stm32l1xx_hal

I have my Pins:

```
    let csPin = gpiob.pb0.into_push_pull_output();
    let busyPin = gpiob.pb8.into_floating_input();
    let sdnPin = gpioa.pa2.into_push_pull_output();
```

But when passing it into the driver:

```
    let mut hal = SpiWrapper::new(spi, csPin, delay);
    hal.with_busy(busyPin.downgrade());
    hal.with_reset(sdnPin.downgrade());
```

`with_busy` and `with_reset` both expect the same type `Output` but `busyPin` is of type `PB0` and ´sdnPin´ is of type `PA2`. Even the downgrade just results in `PA` and `PB` which leads to a compile error.

Is it an issue with the `stm32l1xx_hal` or with the SX driver? Or is there any way I can convert / wrap my types to solve the issue?

ryankurte, 2021-02-14:
hmm, the type bounds should support this (it was a problem some versions ago), it's possible you're battling the type inference but that'd usually give a useful error...

have you tried using [`Sx127x::spi`](https://docs.rs/radio-sx127x/0.10.1/radio_sx127x/struct.Sx127x.html#method.spi) directly? you ideally shouldn't need to construct the `SpiWrapper` yourself.

Niondir, 2021-02-20:
Thanks, using the master did solve the type issue. Now i'm struggling with the SPI traits, see #18

closed 2021-02-20 by Niondir