rust-iot/rust-radio-sx127x #33: example cannot find its own crate

pdgilbert, 2021-03-02:
I was trying to change too much at once and it was shaking my sanity. Before attempting  the compat layer I thought I would go back to a point I had working and try to put my example in my `rust-radio-sx127x` fork. So I branched just before "update everything" and have been trying to add it. But I cannot get the example code to find the modules of the crate it is in. I've tried referring to it as both `radio_sx127x` and as `crate`, and hundreds of other variations with no luck:
```
error[E0433]: failed to resolve: maybe a missing crate `radio_sx127x`?
  --> examples/lora_spi_send.rs:31:5
   |
31 | use radio_sx127x::{prelude::Error, prelude::Sx127x};
   |     ^^^^^^^^^^^^ maybe a missing crate `radio_sx127x`?

error[E0433]: failed to resolve: maybe a missing crate `device`?
  --> examples/lora_spi_send.rs:34:13
   |
34 | use crate::{device::lora::{LoRaConfig, LoRaChannel, Bandwidth, SpreadingFactor, CodingRate,
   |             ^^^^^^ maybe a missing crate `device`?
```
I've also tried adding it to its own Cargo.toml  with no luck, and tried changing the directory name from `rust-radio-sx127x` to `radio-sx127x` just in case there was confusion between the crate name and the directory name, also with no luck.

The example does work outside the `radio_sx127x` crate as reported at https://pdgilbert.github.io/eg_stm_hal/#table-of-additional-examples-status and I have put examples in other crates, where I simply used `use` with the crate name like would be done normally to use it, and no need to add it to Cargo.toml. 

Any suggestions?

pdgilbert, 2021-03-03:
Found the problem. Somehow `edition = "2018"` was missing in the Cargo.toml   package section so `external crate radio-sx127x` was needed in the example code before` use`. (The crate is considered external in the examples.)

closed 2021-03-03 by pdgilbert

ryankurte, 2021-03-03:
> Found the problem. Somehow edition = "2018" was missing in the Cargo.toml package section so external crate radio-sx127x was needed in the example code before use. (The crate is considered external in the examples.)

ooof, that one is terrible / nice find! be good to add that edition keyword 😂

ryankurte, 2021-03-03:
> I was trying to change too much at once and it was shaking my sanity. Before attempting the compat layer I thought I would go back to a point I had working and try to put my example in my rust-radio-sx127x fork.

pros and cons of the type system eh. if you get stuck on things feel do free to keep posting em (or, email if it's not PRable), it's great to know where these abstractions work well and fail to hopefully improve em ^_^