rust-iot/rust-radio-sx127x #44: TX_DONE interrupt not firing.

Cightline, 2021-06-10:
I can't seem to get the TX_DONE interrupt to fire, however the RX_DONE interrupt works fine. 

Example of the problem:
1. Send a packet
2. TX_DONE interrupt *should* fire, but it doesn't, I currently have to poll. 
3. Receive a packet
4. RX_DONE interrupt fires, no polling needed.
5. I can check the bit flags and see that both TX_DONE and RX_DONE are set.

ryankurte, 2021-06-10:
I take it this is via the `DIO0` pin? To receive both it looks like you'll need to switch the `DIO0` mapping in `RegDioMapping1` from `RxDone` (`0b00`) to `TxDone` (`0b01`) between operations (after enabling `RxReady` and/or `TxReady` IRQs in `RegIrqFlags1).

(It'd be neat to have an API for this in the driver if there's a reasonable way of representing it)

[image]

[image]

(from the datasheet [here](https://semtech.my.salesforce.com/sfc/p/#E0000000JelG/a/2R0000001Rbr/6EfVZUorrpoKFfvaF_Fkpgp5kzjiNyiAbqcpqh9qSjE)

Cightline, 2021-06-11:
Very helpful response, and yeah it's `DIO0` pin, I'll try out what you wrote and bring the results back here so we can potentially figure out an API.