rust-iot/rust-radio-sx127x #46: sharing spi?

pdgilbert, 2021-06-24:
In theory I can put a few devices on the `spi` bus. However, I think `lora` consumes `spi`. Am I correct in thinking there may be a problem, or is this something I can try easily?

ryankurte, 2021-06-27:
technically yes one can share an SPI device between peripherals, for bus sharing in general you can use [rahix/shared-bus](https://github.com/Rahix/shared-bus). however, because of the need to manage _chip select_ as well this was unsound so is not currently available for SPI (see https://github.com/Rahix/shared-bus/issues/8) 

https://github.com/rust-embedded/embedded-hal/issues/180 should resolve this but needs following up and I haven't had the time recently.

pdgilbert, 2021-06-28:
Ok. Thanks. I guess I'll put experimenting with this on the back burner for awhile.

closed 2021-06-28 by pdgilbert